JeremyGrosser/rp2040_hal #8: Documentation site

JeremyGrosser, 2021-09-13:
In addition to the READMEs, example code, blog posts, and videos, a documentation site should exist.

# Getting Started
- Introduction - What is this project, who is it for?
- Supported Hardware
- How to get help
- Introduction to toolchains and runtimes
- Setting up Alire
- Connecting a SWD debugger

# Development Guide
- Repository layout
- Hardware overview, links to datasheets
- Clocks
- GPIO
- Timer, RTC, SysTick
- Serial interfaces (UART, SPI, I2C)
- PWM
- ADC
- DMA
- PIO
- Interpolator
- ROM functions
- Low power operation

# API Reference
Make it nice, like the Python standard library docs. Ideally this would be autogenerated, which means we need something akin to docstrings in comments or a way to associate external documentation comments with procedure and function signatures. There are probably existing tools for this.

damaki, 2021-09-25:
FYI there's [GNATdoc](https://docs.adacore.com/live/wave/gps/html/gnatdoc-doc/index.html) for generating HTML documentation from source code comments. It's similar to Doxygen, but doesn't have quite as many features.

JeremyGrosser, 2021-09-26:
I've started working on this, a draft version is available at https://pico-doc.synack.me/

JeremyGrosser, 2021-10-04:
New docs are now live. https://pico-doc.synack.me/

closed 2021-10-04 by JeremyGrosser