JeremyGrosser/rp2040_hal #28: elf2uf2 port

JeremyGrosser, 2022-02-23:
The boot ROM supports flashing using files in UF2 format. Upstream pico-sdk contains a small C program that converts an ELF file to UF2. It would be nice to have this available in Alire so that users don't have to download and build pico-sdk just to flash a board. Maybe there's a clever way to build this utility as a dependency and call it during `alr build`.

NicoPy, 2022-02-24:
I translated elf2uf2 into Ada a few weeks ago, as an exercise. 
I did not publish it since it is not properly tested.
Also, sources have to be cleaned up.

JeremyGrosser, 2022-02-24:
If that's something you'd be willing to contribute, I think the best approach would be to create a separate GitHub repository and Alire crate for it, then we can add it as a dependency to any project that needs to generate uf2 files.

NicoPy, 2022-02-25:
> If that's something you'd be willing to contribute

Sure.

> I think the best approach would be to create a separate GitHub repository

I will.

> and Alire crate for it

Never did this. I'll have to learn how to do this.

Is there a prefered license type for this project ?

JeremyGrosser, 2022-02-25:
If you need help with Alire, the people on gitter are very friendly.
https://gitter.im/ada-lang/Alire

I've been using the BSD-3-Clause license for RP2040 related projects. Most of the official Raspberry Pi code does too. As this will be a separate project, you're free to use whatever license you want.