JeremyGrosser/rp2040_hal #33: I2C Mem_Write sends a repeated start

JeremyGrosser, 2022-02-26:
See conversation on Gitter for details.

hgrodriguez, 2022-02-26:
as requested by Jeremy I am attaching the pictures of the logic analyser
[image]
[image]
[image]

hgrodriguez, 2022-02-28:
I am afraid, that maybe Mem_Read needs fixing, too.

JeremyGrosser, 2022-02-28:
I changed Mem_Write to use a single Master_Transmit rather than a repeated start between address and data (5879969b90cb0890e3ab445f598431037e00973c).

Mem_Read needs the repeated start because it has to change the I2C address' LSB from write to read between sending the memory address and reading the data.

[image]
[image]

closed 2022-02-28 by JeremyGrosser