JeremyGrosser/rp2040_hal #44: Default XOSC_Startup_Delay

Fabien-Chouteau, 2022-07-14:
Hi @JeremyGrosser,

I got an issue with the default `XOSC_Startup_Delay` on a Pico board.

My firmware locked in the `RP.Clock.Initialize` but only on a "cold" start, so I looked a the startup delay.
Looking at the `pico-sdk` the startup delay is computed this way:
```c
#define STARTUP_DELAY (((((XOSC_MHZ * MHZ) / 1000) + 128) / 256) * PICO_XOSC_STARTUP_DELAY_MULTIPLIER)
```
For 12Mhz, the formula gives:
```
((XOSC_MHZ * MHZ) / 1000) + 128) = ((12000000 / 1000) + 128) = 12128
```
which is close to the default in `rp2040_hal` which is `12_032`.
https://github.com/JeremyGrosser/rp2040_hal/blob/6115fd1ee50c2de117082a46c9ba3bd9da967c14/src/drivers/rp-clock.ads#L21

However there is this `PICO_XOSC_STARTUP_DELAY_MULTIPLIER` value that is set to `64` for different boards (all the Adafruit ones and the seeed xiao). I changed the `XOSC_Startup_Delay` to `12_032 * 64` and now the board boots flawlessly.

So I don't know what's going on with this specific Raspberry Pi Pico. Maybe out of spec crystal or capacitors. Maybe my carrier board is having an effect...

I have a way to fix the issue on my side, but I wonder if we should use a "safer" default for the `XOSC_Startup_Delay` value.

JeremyGrosser, 2022-07-14:
`12_032` was chosen to be roughly 1 millisecond with a 12 MHz crystal. 64 milliseconds seems extreme, is there any value in between that works for you?

Fabien-Chouteau, 2022-07-14:
I took 64 because that's the value used for the Adafruit boards. In theory we don't have to do this but it's a nasty issue to deal with.