JeremyGrosser/rp2040_hal #47: I2C HAL.Mem_Read always times out

NathanRignall, 2022-10-07:
When testing the example code in pico_examples for I2C using a BMP280 the low level implementation of the I2C driver works but the abstracted HAL driver does not. It seems to get stuck on Read_Ready when in read stage of the read and subsequently time out.

This was tested with a pair of BMP280s and a Pi Pico.
Also tested with a BMP384 on custom rp2040 board which produces the same result.

NathanRignall, 2022-10-07:
When using 7 bit addresses the RP.I2C_Master.Set_Address shifts the address one bit to the right.

This results in the address in the low level RP driver to be set to 111011 compared to 1110110 when the low level driver is called directly in the example.

Removing the bit shift in RP.I2C_Master.Set_Address fixes the issue for 7 bit addresses and the example read works as expected.

JeremyGrosser, 2022-10-07:
This is expected behavior. All of the other HAL.I2C implementations do this same shift of the address, so RP.I2C_Master needs to as well, or the device drivers in Ada_Drivers_Library won't work.

NathanRignall, 2022-10-07:
Ah okay sorry, I didn't realise, it's very strange!

Should the example be updated for this then to use the different address?

JeremyGrosser, 2022-10-07:
I agree, this behavior is confusing. This was a major motivation for the RP.I2C driver to use UInt7 and UInt10 for addresses.

Yes, we should update the example.

NathanRignall, 2022-10-07:
Thanks, shall I update the example? I've not contributed to a projects repo on Github yet and would be fun to try :)

I don't know much about 10 bit i2c addressing but if the I2C_Address type is using all 10 bits the user cannot shift their address beforehand?

JeremyGrosser, 2022-10-07:
There's a (somewhat stale) conversation about potentially making some breaking changes to the `HAL.I2C` interface in https://github.com/AdaCore/Ada_Drivers_Library/issues/401

I prefer an interface where users don't have to do any shifting at all and there's a clear distinction between addressing modes, which is what I implemented with `RP.I2C`.

If you want to send a PR for the example, that saves me some work, so go for it.

NathanRignall, 2022-10-07:
Interesting read, thanks... hope the issue gets somewhere! `RP.I2C` is very clear it allowed me to understand what was going on in the `HAL.I2C` interface.

Yes I agree, it is very counter intuitive. In fact this possibly why I never got your original implementation of the I2C driver working and just never realised.

Thanks, I'll get that in.

closed 2022-10-18 by JeremyGrosser