JeremyGrosser/rp2040_hal #48: 10-bit I2C addresses are split into two writes

JeremyGrosser, 2022-11-14:
Reported by @hgrodriguez on gitter:

When calling `Mem_Read` with `Memory_Size_16b`, the two address bytes are written in separate transactions, rather than using repeated start.

[image]

JeremyGrosser, 2022-11-14:
I'm not able to reproduce this issue. Here's the test code I'm using:

```ada
with RP.Device;
with RP.Clock;
with RP.GPIO;
with RP.I2C_Master;
with Pico;
with HAL.I2C;
with HAL; use HAL;

procedure Eeprom is
   use RP.GPIO;
   use HAL.I2C;

   SCL  : GPIO_Point renames Pico.GP15;
   SDA  : GPIO_Point renames Pico.GP14;
   Port : RP.I2C_Master.I2C_Master_Port renames RP.Device.I2CM_1;
   Addr : constant I2C_Address := 2#1010_000_0#;

   Data   : I2C_Data (1 .. 32) := (others => 1);
   Status : I2C_Status;
begin
   RP.Clock.Initialize (Pico.XOSC_Frequency);
   RP.Clock.Enable (RP.Clock.PERI);
   Pico.LED.Configure (Output);
   Pico.LED.Clear;

   SCL.Configure (Output, Pull_Up, RP.GPIO.I2C, Schmitt => True);
   SDA.Configure (Output, Pull_Up, RP.GPIO.I2C, Schmitt => True);
   Port.Configure (Baudrate => 400_000);

   RP.Device.Timer.Enable;

   loop
      Data := (others => Data (1) + 1);
      Port.Mem_Write
         (Addr          => Addr,
          Mem_Addr      => 0,
          Mem_Addr_Size => Memory_Size_16b,
          Data          => Data,
          Status        => Status,
          Timeout       => 1000);
      RP.Device.Timer.Delay_Milliseconds (5);
      Port.Mem_Read
         (Addr          => Addr,
          Mem_Addr      => 0,
          Mem_Addr_Size => Memory_Size_16b,
          Data          => Data,
          Status        => Status,
          Timeout       => 1000);
      RP.Device.Timer.Delay_Milliseconds (5);
      if Status = Ok then
         Pico.LED.Set;
      end if;

      RP.Device.Timer.Delay_Seconds (1);
   end loop;
end Eeprom;
```

The `Mem_Read` transaction looks like this:
[image]

hgrodriguez, 2022-11-15:
I hope we are using the very same version (from my alire.toml):
[[depends-on]]
pico_bsp = "^2.0.0"

hgrodriguez, 2022-11-15:
I do confirm, that your example works, therefore please close the ticket, looks like my code is not OK. Sorry to waste your time. BTW: I like your logic analyzer view, which model is this?