JeremyGrosser/rp2040_hal #49: I2C: 16bit EEPROM dresses are split on the SECOND read

hgrodriguez, 2022-11-17:
I2C: 16bit EEPROM dresses are split on the SECOND read

Hi Jeremy,
I spent some time investigating my code and could not find the problem.
Then I came back to your code and adjusted it a little bit to enable me to debug it better.
I do appreciate, that you do not have time to resolve this soon, I just wanted to make sure that I used your code (not mine) to reproduce it.
Let me know, if I did something wrong.

I have this in my alire.toml:
[[depends-on]]
pico_bsp = "^2.0.0"

See code below:

with RP.Device;
with RP.Clock;
with RP.GPIO;
with RP.I2C_Master;
with Pico;
with HAL.I2C;
with HAL; use HAL;

procedure Jeremy is
   use RP.GPIO;
   use HAL.I2C;

   SCL  : GPIO_Point renames Pico.GP15;
   SDA  : GPIO_Point renames Pico.GP14;
   Port : RP.I2C_Master.I2C_Master_Port renames RP.Device.I2CM_1;
   Addr : constant I2C_Address := 2#1010_000_0#;

   --------------------------------------------------------------------------
   --  HOLGER: make the size of the array adjustable
   ARRAY_SIZE : constant Integer := 1;

   --------------------------------------------------------------------------
   --  HOLGER: add array for Writing
   Data_W     : I2C_Data (1 .. ARRAY_SIZE) := (others => 0);
   
   --------------------------------------------------------------------------
   --  HOLGER: add array for Reading
   Data_R     : I2C_Data (1 .. ARRAY_SIZE);
   
   Status     : I2C_Status;
begin
   RP.Clock.Initialize (Pico.XOSC_Frequency);
   RP.Clock.Enable (RP.Clock.PERI);
   Pico.LED.Configure (Output);
   Pico.LED.Set;

   SCL.Configure (Output, Pull_Up, RP.GPIO.I2C, Schmitt => True);
   SDA.Configure (Output, Pull_Up, RP.GPIO.I2C, Schmitt => True);
   Port.Configure (Baudrate => 400_000);

   RP.Device.Timer.Enable;

   --------------------------------------------------------------------------
   --  HOLGER: complete loop unrolling to focus on the problem
   --          and make debugging easier
   
   --------------------------------------------------------------------------
   --  HOLGER: Case Data_W (1) := 1
   Data_W := (others => Data_W (1) + 1);
   Port.Mem_Write
     (Addr          => Addr,
      Mem_Addr      => 0,
      Mem_Addr_Size => Memory_Size_16b,
      Data          => Data_W,
      Status        => Status,
      Timeout       => 1000);
   if Status = Ok then
      Pico.LED.Set;
   end if;
   RP.Device.Timer.Delay_Milliseconds (500);
      
   Port.Mem_Read
     (Addr          => Addr,
      Mem_Addr      => 0,
      Mem_Addr_Size => Memory_Size_16b,
      Data          => Data_R,
      Status        => Status,
      Timeout       => 1000);
   if Status = Ok then
      Pico.LED.Set;
   end if;
   RP.Device.Timer.Delay_Milliseconds (500);
      
   --------------------------------------------------------------------------
   --  HOLGER: compare the array written with the array read
   for I in 1 .. ARRAY_SIZE loop
      if Data_W (I) /= Data_R (I) then
         --------------------------------------------------------------------
         --  Holger: STOP if data different: it is the same
         Pico.LED.Clear;
         loop
            null;
         end loop;
      end if;
   end loop;

   --------------------------------------------------------------------------
   --  HOLGER: Case Data_W (1) := 2
   Data_W := (others => Data_W (1) + 1);
   Port.Mem_Write
     (Addr          => Addr,
      Mem_Addr      => 0,
      Mem_Addr_Size => Memory_Size_16b,
      Data          => Data_W,
      Status        => Status,
      Timeout       => 1000);
   if Status = Ok then
      Pico.LED.Set;
   end if;
   RP.Device.Timer.Delay_Milliseconds (500);
      
   Port.Mem_Read
     (Addr          => Addr,
      Mem_Addr      => 0,
      Mem_Addr_Size => Memory_Size_16b,
      Data          => Data_R,
      Status        => Status,
      Timeout       => 1000);
   if Status = Ok then
      Pico.LED.Set;
   end if;
   RP.Device.Timer.Delay_Milliseconds (500);
      
   --------------------------------------------------------------------------
   --  HOLGER: compare the array written with the array read
   for I in 1 .. ARRAY_SIZE loop
      if Data_W (I) /= Data_R (I) then
         --------------------------------------------------------------------
         --  Holger: STOP if data different -> it *is* different!
         Pico.LED.Clear;
         loop
            null;
         end loop;
      end if;
   end loop;

   --------------------------------------------------------------------------
   --  HOLGER: we end here if everything is OK
   loop
      Pico.LED.Set;
   end loop;
   
end Jeremy;

I used GDB to step through the program, but the same behaviour is true if run as normal without debugger attached.

Those are the results:
Write Data=1 -> OK
[image]

Read Data=1 -> OK
[image]

Write Data=2 -> OK
[image]

Read Data=2 -> NOK
[image]

JeremyGrosser, 2022-11-18:
I have a fix, but I'm afraid it might break other I2C devices. Please test out the https://github.com/JeremyGrosser/rp2040_hal/tree/issue_49 branch and let me know if it solves your problem.

hgrodriguez, 2022-11-19:
Thanks I will look at it as early as I can. Currently my work bench has another project so maybe next weekend

hgrodriguez, 2022-11-26:
i checked and the behaviour did not improve, same separation of the address still there. When I remember correctly, pico_bsp 1.6.0 worked fine with 16 bit addresses, not sure what you changed in the i2c code. I will try to spend more time, but at the moment I have a different project going.

hgrodriguez, 2022-12-14:
I had some time to debug in the libraries. I am not sure, if this helps, what I write below.
My view is, that if I call a procedure multiple times with the very same parameters, then the behaviour should be the same.
I found the following using the code above.

The first time I call Port.Mem_Write on line 49, I found out, that in the procedure RP.I2C.Write the flags for:
```
P.IC_DATA_CMD :=
         (RESTART => (if This.Repeated_Start and P.IC_DATA_CMD.FIRST_DATA_BYTE = ACTIVE then ENABLE else DISABLE),
          STOP    => (if not This.Repeated_Start and This.TX_Remaining = 1 then ENABLE else DISABLE),
          CMD     => WRITE,
          DAT     => Data,
          others  => <>);
```
result in: 
```
P.IC_DATA_CMD :=
         (RESTART => DISABLE,
          STOP    => DISABLE,
          CMD     => WRITE,
          DAT     => Data,
          others  => <>);
```

The second time I call Port.Mem_Write on line 89, I found out, that in the procedure RP.I2C.Write the flags for:
```
P.IC_DATA_CMD :=
         (RESTART => (if This.Repeated_Start and P.IC_DATA_CMD.FIRST_DATA_BYTE = ACTIVE then ENABLE else DISABLE),
          STOP    => (if not This.Repeated_Start and This.TX_Remaining = 1 then ENABLE else DISABLE),
          CMD     => WRITE,
          DAT     => Data,
          others  => <>);
```
result in: 
```
P.IC_DATA_CMD :=
         (RESTART => ENABLE,
          STOP    => DISABLE,
          CMD     => WRITE,
          DAT     => Data,
          others  => <>);
```

As written above, not sure if this helps, but even having the very same parameters in both calls, I would not expect, that those flags are different.
Maybe this is a hint for you.

hgrodriguez, 2022-12-14:
I would like to offer to send you EEPROMS like the I use:
https://www.mouser.com/ProductDetail/Microchip-Technology/24FC64F-I-P?qs=WqWCsLCZBkoe3htI4nCQOg%3D%3D
and ship it to your private address if this helps to debug. Let me know.

JeremyGrosser, 2024-10-24:
I believe I've fixed the behavior of `RP.I2C_Master` with release 2.4.0. Sorry it took so long.

hgrodriguez, 2025-05-17:
Hi Jeremy,
I had time (I know it is a while ago) to check this again.

This is my alire.toml
name = "example_pico"
description = "Examples for EEPROM_I2C crate"
version = "1.0.0"
licenses = "BSD-3-Clause"

authors = ["Holger Rodriguez"]
maintainers = ["Holger Rodriguez <github@roseng.ch>"]
maintainers-logins = ["hgrodriguez"]
tags = ["embedded", "rp2040", "eeprom", "i2c"]
website = "https://github.com/hgrodriguez/eeprom_i2c_examples"

executables = ["example_pico"]

[gpr-set-externals]
#  I want all dependencies with full debug information
ATOMIC_BUILD_MODE="debug"
BBQUEUE_BUILD_MODE="debug"
CORTEXM_BUILD_MODE="debug"
EDC_CLIENT_BUILD_MODE="debug"
EEPROM_I2C_STYLE_CHECKS="enabled"
EEPROM_I2C_BUILD_MODE="debug"
ITSYBITSY_RP2040_BSP_BUILD_MODE="debug"
PICO_BSP_BUILD_MODE="debug"
RP2040_HAL_BUILD_MODE="debug"
USB_EMBEDDED_BUILD_MODE="debug"

[[depends-on]]  # Added by alr
gnat_arm_elf = "^14"  # Added by alr

[[depends-on]]
eeprom_i2c = "6.0.1"

[[depends-on]]
pico_bsp = "2.2.0"

[[depends-on]]
ltp_305 = "0.3.0"

[build-profiles]
"*" = "development"
so I updated to the latest pico_bsp.

Unfortunately, the very same code as above, produces the very same error, that the last read split the 16 bit address and therefore it does not work.

I do understand, that you have other priorities and your bench is quite probably busy with another project(s).
Nevertheless, if you do not have time, can you point me in any direction where I can investigate/debug more so that this can be fixed?
I now have a project here, which would highly benefit from a 16 bit I2C address working properly.

Thanks for any help.

JeremyGrosser, 2025-05-17:
I don’t have a solution for you. I’ve run into several situations with the hardware I2C on RP2040 where the state machine doesn’t do the right thing but I don’t understand why. It might be a bug in the driver, but I’ve not been able to find a fix that doesn’t break something else.

For my own projects, I’ve given up on hardware I2C and bit banged a software implementation: https://github.com/JeremyGrosser/softdev

I cannot provide support for that library either. There are simply too many ways for I2C to misbehave for me to anticipate any problems you might have. You’re going to have to debug it yourself.

hgrodriguez, 2025-05-18:
I appreciate your answer and thanks a lot for your effort!
Would this work for 16bit EEPROM addressing:
C_Memory_Address_Size => HAL.I2C.Memory_Size_16b
?

JeremyGrosser, 2025-07-22:
I have rewritten RP.I2C_Master again, following the reference implementation from pico-sdk very closely. I wrote unit tests for I2C_Master that exercise the Mem_Read and Mem_Write procedures extensively on real hardware with a [Microchip 24FC64F](https://www.digikey.com/en/products/detail/microchip-technology/24FC64F-I-P/2665915) and [CAT24C32WI-GT3](https://www.digikey.com/en/products/detail/onsemi/CAT24C32WI-GT3/1631007). These tests all pass for me.

A few notes about the test suite:
- 8-bit tests are commented out, I don't have an 8-bit EEPROM to test with.
- Setting the Baudrate to 1 MHz (`1_000_000`) results in an actual SCL clock of 844 KHz. This is a limitation of the RP2040 hardware. This is still acceptable for the I2C High Speed standard, the timings are specified as a maximum.
- Coverage report shows that some of the error handling for timeouts are untested. This would require a more sophisticated fault injection setup to test.

These changes are now on the master branch. Please pin to that branch and verify that it works on your hardware. Assuming that it does, I will tag a release.

hgrodriguez, 2025-07-22:
thanks for this, I will try it next week, this week I do not have time.

hgrodriguez, 2025-07-28:
Hi Jeremy,
fantastic job!

I hereby confirm, that the current master of rp2040_hal works with *all* EEPROMs under test:
https://github.com/hgrodriguez/eeprom_i2c_examples/tree/main/examples_rp2040

Thank you so much!
I do appreciate it!

Regards,
Holger

closed 2025-07-29 by JeremyGrosser

hgrodriguez, 2025-07-30:
Will you update pico_bsp to use the new version?
Thanks!

JeremyGrosser, 2025-07-30:
pico_bsp should use any version starting with 2. Use “alr index —update-all; alr update” to make Alire pull the new rp2040_hal.