JeremyGrosser/rp2040_hal #55: Calling reset_to_usb_boot does not properly reset the device

ThyMYthOS, 2024-01-25:
A call to `reset_to_usb_boot` somehow locks up the device, but it will not properly reboot and show up on USB. Maybe some preparation is required, but looking at the pico-sdk I could not find anything special.

https://github.com/JeremyGrosser/rp2040_hal/blob/02c3d00151ebe62777ffe9a525d36d7f9473aa2b/src/drivers/rp-rom.ads#L154

ThyMYthOS, 2024-01-26:
The function should be marked as "No_Return", but this does not fix the issue.

JeremyGrosser, 2024-01-28:
I am able to reproduce this issue. `reset_to_usb_boot` does work if called before `RP.Clock.Initialize`, but not after.

I think the [_usb_boot](https://github.com/raspberrypi/pico-bootrom/blob/master/bootrom/bootrom_main.c#L184) and subsequent [_usb_clock_setup](https://github.com/raspberrypi/pico-bootrom/blob/master/bootrom/bootrom_main.c#L111) procedures in the bootrom are expecting the clocks to be in a specific state before entry and we don't have an easy way to get back to the default state after `RP.Clock.Initialize`

ThyMYthOS, 2024-01-30:
@JeremyGrosser thanks for the hint. `pico-sdk` does not disable the `ROSC`. So if we remove line 127 in `rp-clock.adb` or if `ROSC` is re-enabled right before the call to `reset_to_usb_boot` the issue is fixed.

https://github.com/JeremyGrosser/rp2040_hal/blob/b629518ee7b44d14a246b533a0eda6469fe67dd8/src/drivers/rp-clock.adb#L127

commit f5877d7b (2024-01-31): Don't disable ROSC after switching to XOSC in RP.Clock.Initialize. Fixes #55
closed 2024-01-31 by JeremyGrosser